GridSpace/grid-apps #91: Add a ruler for the grid

juodumas, 2020-10-27:
A ruler would be useful to check part dimensions before exporting.

This proof of concept kind of works:

```javascript
diff --git a/src/moto/space.js b/src/moto/space.js
index 06589e0..6711326 100644
--- a/src/moto/space.js
+++ b/src/moto/space.js
@@ -248,13 +248,29 @@
             h = y / 2,
             d = z / 2,
             zp = -d - platformZOff + gridZOff,
-            majors = [], minors = unitMinor ? [] : null, i;
+            majors = [], minors = unitMinor ? [] : null, i,
+            labelSize = gridUnitMinor * 1.5,
+            canvasScale = 4,
+            canvasPadding = 10,
+            xLabelsMesh,
+            xLabelCanvas = document.createElement('canvas'),
+            xLabelCtx = xLabelCanvas.getContext('2d');
+
+        xLabelCanvas.width = x * canvasScale + canvasPadding * 2 * canvasScale;
+        xLabelCanvas.height = labelSize * canvasScale;
+        xLabelCtx.scale(canvasScale, canvasScale);
+        xLabelCtx.fillStyle = '#333333';
+        xLabelCtx.font = labelSize + 'px monospace';
+        xLabelCtx.textAlign = 'center';
+        xLabelCtx.textBaseline = 'top';
+
         for (i = -xo; i <= xo; i += unitMinor) {
             let oh = isRound ? Math.sqrt(1-(i/xo)*(i/xo)) * h : h,
                 dM = Math.abs(i % unitMajor);
             if (i < -w || i > w) continue;
             if (dM < 1 || Math.abs(unitMajor - dM) < 0.1) {
                 majors.append({x:i, y:-oh, z:zp}).append({x:i, y:oh, z:zp});
+                xLabelCtx.fillText('' + (i + w), i + xo + canvasPadding, 0)
             } else {
                 minors.append({x:i, y:-oh, z:zp}).append({x:i, y:oh, z:zp});
             }
@@ -273,6 +289,18 @@
         if (minors) gridView.add(makeLinesFromPoints(minors, gridColorMinor || 0xcccccc, 1));
         if (oldGridView) Space.scene.remove(oldGridView);
         Space.scene.add(gridView);
+ (grid labels for x & y axis labels)
+        xLabelsMesh = new THREE.Mesh(
+            new THREE.PlaneGeometry(x + canvasPadding * 2, labelSize),
+            new THREE.MeshBasicMaterial({
+                transparent: true,
+                map: new THREE.CanvasTexture(xLabelCanvas),
+            })
+        );
+        xLabelsMesh.position.x = 0;
+        xLabelsMesh.position.y = - h - labelSize;
+        xLabelsMesh.position.z = zp;
+        gridView.add(xLabelsMesh);
     }
 
     function setOrigin(x, y, z) {
```

[image]

AlbertMN, 2020-10-28:
Cool idea for a setting! Would surely like to see this implemented :)

stewartoallen, 2020-10-28:
Agree. I'll look at adding this as an option in preferences. Thanks.

juodumas, 2020-10-28:
I could also try to do a PR based on the proof of concept above if it would help. Just not sure if I know all the things that depend on this code in `space.js`.

stewartoallen, 2020-10-28:
I would happily accept a PR that does this with the following modifications

- adds an API setter endpoint on space.js `setRulers(x,y,center)`
- x, y, and center are booleans
- x, y control visibility of rulers
- center allows for origin center configurations

this should be toggle-able in real time. shouldn't be a problem if you keep the code in `setGrid` (yes, that code is old and ugly)

This will probably look odd with round beds. I will add a UI preference that drives the new API.

juodumas, 2020-10-28:
I didn't even know that circular beds are supported. Only ever used the laser mode. The labels don't look too bad:
[image]

> center allows for origin center configurations

What should change when center origin is used? The code posted above seems to work the same with center origin:
[image]

> shouldn't be a problem if you keep the code in `setGrid`

Do you mean the label drawing code can stay in `setGrid()`? The advantage is that the cycle that draws the grid is re-used, the disadvantage is that I will have to call setGrid from `setRulers()` and also remember setGrid's parameters...

stewartoallen, 2020-10-28:
delta 3D printers use round beds. in the case of origin center, I would like the 0 to be in the middle with numbers to the left being negative.

stewartoallen, 2020-10-28:
just as happy to have the setRulers code entirely independent of setGrid. setGrid would call setRulers but not the other way around. grid affects rulers, not the other way around.

juodumas, 2020-10-28:
Makes sense. Will see what I can do.

stewartoallen, 2020-10-29:
thanks for the pull request.

closed 2020-10-29 by stewartoallen

juodumas, 2020-10-29:
Thanks for this project and for merging. This will be useful to us.