GridSpace/grid-apps #93: How to solve "Cannot get /kiri"?

tovrstra, 2020-11-28:
I tried a local install as follows on Fedora Server 33, starting in the home directory of an normal user:

```bash
# Install NVM
git clone https://github.com/creationix/nvm.git ~/.nvm
(cd ~/.nvm; git checkout `git describe --abbrev=0 --tags`)
. ~/.nvm/nvm.sh
nvm install 14
nvm use 14
# Install GridSpace apps
npm install -g @gridspace/app-server
gs-app-server --debug
```

Then opening http://localhost:8080/kiri just shows one line `Cannot GET /kiri`. I'm have little experience with node js, so my apologies in case this is a silly question. How can this be fixed? Thanks!

stewartoallen, 2020-11-28:
are you in the root of the [grid-apps](https://github.com/GridSpace/grid-apps) source tree when you run that command?

tovrstra, 2020-11-29:
@stewartoallen Thanks for your comment! Now I understand how it works. I'm adding my updated steps below in case they could of help to other n00bs:

```bash
# Install NVM
git clone https://github.com/creationix/nvm.git ~/.nvm
(cd ~/.nvm; git checkout `git describe --abbrev=0 --tags`)
. ~/.nvm/nvm.sh
nvm install 14
nvm use 14
# Install GridSpace apps
git clone https://github.com/GridSpace/grid-apps.git
cd grid-apps
npm i  # installs dependencies listed in packages.json
npm install -g @gridspace/app-server
gs-app-server --debug
```

It might be helpful to add the `cd grid-apps` the instructions in the `readme.md` file.

PR #94 (2020-11-29): Improve local testing instructions in readme.md
Fixes #93
closed 2020-11-29 by stewartoallen