GridSpace/grid-apps #96: TypeError: i[e] is undefined when slicing contour

jsiddall, 2020-11-30:
I did something on my local Kiri:Moto that is causing a "TypeError: i[e] is undefined" message to pop up on the client when slicing starts the contour step.  I have sliced this same part successfully in the past so not sure what changed.

Two questions:
1. Any ideas how to resolve
and
2. Assuming no to question 1, how can I get a little more information to help with debugging?

stewartoallen, 2020-11-30:
if you are working against the very latest code base, it's possible a recent commit broke something. since I'm pushing new code hourly, try again and see if it's fixed. otherwise, try rolling back to the last known working commit.

if you want to be able to debug better, then run the server-side with the `debug` option like this:

`node app-server --debug`

when you encounter errors, you will get a clear stack trace.

jsiddall, 2020-12-01:
Yup, spot on.  2.4.D7 pulled Nov 29 14:29 slices the same part fine, so it looks like something happened in 2.4.D8.

However, I also tried adding the debug option on while running 2.4.D8.  I couldn't get node to run app-server as you had in your post, with some error about app-server not found, but I had been successfully running from the gs-app-server script so I just added --debug to that:

```
/usr/local/grid-apps# gs-app-server --debug --port 8082
201130.201720 '[grid]' 'creating shared kvstore'
201130.201720 '[appserver]' 'initialized grid from apps/grid'
201130.201720 '[appserver]' 'app-server running: ports=[8082] apps=[apps] data=[data] logs=[logs]'
```

But even after restarting that way I only see the 3 lines above, after getting the error on the client.  The logs don't contain anything either.  Strangely, I am getting a different error now when slicing:

TypeError: buckets[j] is undefined

To be clear, the other slices and animations I was posting from today in the Kiri:Moto 2.4 Roadmap and Updates forum thread were also done with 2.4.D8 so not all parts are experiencing contour slicing issues.  Any other thoughts on how I can get you something useful for debugging?

stewartoallen, 2020-12-01:
is this *still* the case with the very latest pull? I fixed a slicing "bucket" issue earlier today. If need your model and settings to replicate.

jsiddall, 2020-12-01:
> is this _still_ the case with the very latest pull?

Hah, no!  I just pulled the latest (now reported as 2.4.D9) and the issue is gone.  Not only that but I see a bunch of other changes too, including a significant speedup in the animation (actually, the whole UI is snappier), the "dot" on the buttons for each of the enable-able actions, plus the XYZ coordinates of the tool.  Nice!

With most projects pulling software earlier in the day is good enough for troubleshooting.  I'm not used to this warp speed development cycle!  Thanks again for all the great code and great support.

closed 2020-12-01 by jsiddall