GridSpace/grid-apps #100: inconsisten "step over" configuration

foxharp, 2020-12-17:
For some reason, "step over" in "roughing" can be configured down to .01 (1%), but in contouring it can only go down to .05.  In both cases, the tooltip implies the minimum is .1.   IMO it should be .01 in all cases.

And of course, there's no "step over" setting in outlining at all.  As long as there's any possibility of overlapping cuts (e.g., wide cutouts, or some interaction between inside and outside cuts?) there should be a "step over" setting.   (I know you've talked about removing wide cutouts in the past.)

paul

stewartoallen, 2020-12-17:
inconsistency resolved. true, outlining has no step over. wide cutout uses 50%. this brings up a design issue. it's possible to just throw more parameters at everything. or perhaps it makes sense to have a tool profile library where you define feeds, speeds, step over, plunge rate. then instead of selecting a tool, you select a tool profile (could be tied to a material or other scenario). this would eliminate a ton of parameters on each operation, but allow adding more parameters per tool profile.

thoughts?

foxharp, 2020-12-17:
Perhaps.    I guess it depends on your target "market".  A hobbyist like me just wants to set a few parameters for the current project, then move on to the next one, maybe referring back to previous ones to see what was done.  A professional trying to run an ultra-efficient shop would have different needs.

My spindle speed is manual, with a dial on the the router itself.  My feed rates are all so slow it hardly matters what I set them to.  I used to have issues with burning out bits, so step over was important, but I haven't had that problem in a while (I mostly stopped using 1/8" bits), so it's not that critical either.  So take my requests in this whole area with a grain of salt.  :-)

paul

stewartoallen, 2020-12-17:
step over added to outlining for next release when "wide cutout" enabled

closed 2020-12-17 by stewartoallen