GridSpace/grid-apps #101: Feature request - Kiri:Moto - CNC Mode - Rough Operation Clear faces only

Type8, 2020-12-19:
Hi, 
I know you are thinking to change something about the rough operation. I read about it somewhere in the forum.

Anyway:
basically I do 99% of times 2.5D cam operations and I never use the rough operation actually.
Today I had the need to clear some faces and I saw that this functionality is available only in the ROUGH op, so I added the rough op to the milling operations array (I am using the 2.5 version of Kiri:Moto).

What happened is that the rough operation was applied to all the piece and not only to the faces. What I am trying to say is that I don't think I need for my purpose to rough all the lines and then do the outline operation, I only need to "clear the faces".

Do you think it is possible to implement it?
I could try to do it the christmas vacation if you think it is interesting but you don't have time :)

Umberto.

stewartoallen, 2020-12-19:
I will be separating facing and pocketing operations very soon. The way I'm going to do it, though, is more like the way trace selection is done. IOW, you will select a region (including entire part shadow/boundary) for an operation and then specify tooling and other parameters.

Type8, 2020-12-21:
PERFECT!
Thanks :)

stewartoallen, 2020-12-26:
Please check out [this new feature in 2.5](https://youtu.be/-QUSEBNJiFQ) which should provide what you're looking for.

closed 2020-12-26 by stewartoallen