GridSpace/grid-apps #104: "contour" operation doesn't preserve all settings across delete/repopulate

foxharp, 2020-12-28:
I think you said in the video that the "bottom row" operations in 2.5 all kept their settings, if you delete them and then add them back.

This seems to be true, except for the X/Y dropdown under "contour".  It gets reset to 'X' if the operation is deleted then re-added.

[ Also:  I know the idea is that operations can be run in any order, but there's a natural order in which they'll usually happen.  A heuristic that kept them in that order when first "Added" would probably be right most of the time.  (E.g. there's no reason roughing would ever be last.  It will be first 99% of the time.)

paul

stewartoallen, 2020-12-28:
true about contour. at present, it always adds X first and if an X contour operation is present, it turns it into a Y. I'll look at a default order heuristic. thanks.

stewartoallen, 2021-03-04:
I'm going to close this because this was the designed and intended behavior. Axis X/Y is a parameter of the contour op, not a class of operation unto itself. And while it is possible to have seemingly non-sensical operation ordering, there is a subset of parts with enough limits and constraints that this could be useful under in some scenario. I don't think there is significant burden to dragging the roughing op to the left if it were added later for some reason.

closed 2021-03-04 by stewartoallen