GridSpace/grid-apps #110: Uncaught TypeError: Cannot read property 'round' of undefined

ryans1224, 2021-02-24:
```
Uncaught TypeError: Cannot read property 'round' of undefined
    at Object.i.export (worker.js?2.6.3:1)
    at export (worker.js?2.6.3:1)
    at self.onmessage (worker.js?2.6.3:1)
```

[image]

I _think_ this occurs when pressing 'export' before slicing operation is complete. Maybe worth disabling the export function/button until slicing operation is finished?

Here is relevant workspace:
[workspace (1).zip](https://github.com/GridSpace/grid-apps/files/6034842/workspace.1.zip)

ryans1224, 2021-02-24:
I am trying to replicate, but unable to at this moment in time

ryans1224, 2021-02-24:
Here is  a replication:

Flow: `Preview > Immediately press export`
 
https://user-images.githubusercontent.com/7398104/108983043-5b09db80-7686-11eb-8e10-9fda89247119.mp4

I realise i should wait until the model is sliced to export, however its just something i run into and thought worth logging.

stewartoallen, 2021-02-25:
there was a patch in 2.7 that addresses slice cancelation. I was not able to duplicate the above behavior. can you try 2.7 and see if it's better behaved now?

stewartoallen, 2021-02-27:
is this still happening for you in 2.7?

ryans1224, 2021-03-01:
All fixed in 2.7, just tried 5+ times and couldn't replicate :)

closed 2021-03-01 by ryans1224