GridSpace/grid-apps #111: in CNC mode can not select some lines to trace

GR0B, 2021-02-24:
Reference: https://forum.grid.space/t/what-is-the-correct-way-to-bevel-an-edge-when-cncing/270/4
[kiri.zip](https://github.com/GridSpace/grid-apps/files/6039407/kiri.zip)

When trying to trace (to bevel or V groove) some lines, some lines can be selected but some can not be selected. 
Tested with v2.7 dev on 2021-02-24 
ZIP file contains STL files and annotated screenshots.

Process to replicate (can not select to V groove)
 import file AB test*.stl
 add process Trace, and select lines

Process to replicate (can not bevel edge)
 import files 18mm ext viewing*.stl (2 files)
 lay files flat on bed (ctrl+click face)
 add process Trace, and select side lines to bevel (1 file works but the other does not)

stewartoallen, 2021-02-24:
The reference link doesn't appear to be a zip file. If you're not able to attach it to this ticket, can you drop it in Discord or email sa@grid.space?

GR0B, 2021-02-25:
Github is showing the reference URL (link to forum) and ZIP differently to how I expected. They are 2 different links, you can access the zip by clicking on "Kiri.zip"

stewartoallen, 2021-03-01:
What program generated the STL files? In the first instance, the un-selectable line has Z vertex positions which are not co-planar. So it does not appear "flat" to the slicer.

The same appears true for the second. And I am seeing a lot of vertexes which are slightly mis-aligned. They appear to be off by some floating point error and then scaled which is making it worse.

GR0B, 2021-03-02:
Hi Stewart, 
The parts are designed and exported from Onshape with the medium quality setting. Setting the quality to course sometimes helps with some files but causes other issues.

GR0B, 2021-03-02:
If it is helpful I can share one of the affected designs with you on Onshape so you can see it before exported.

stewartoallen, 2021-03-02:
Certainly. You can share to sa@grid.space inside Onshape. That could help.

commit 50d95e60 (2021-03-04): use entirely different selector for single line traces. fixes #111
closed 2021-03-04 by stewartoallen