GridSpace/grid-apps #113: complex STLs super slow to slice, kiri:moto eventually crashes with STATUS_BREAKPOINT

oliof, 2021-02-26:
Giving kiri:moto a test drive, and while it's doing well with simple STLs, complex STLs like larger scale models with a lot of detail are a challenge. An STL that slices reasonably well in KISSlicer and Simplify3D, slices very slowly in kiri:moto and finally (after more than one hour), the tab crashes with STATUS_BREAKPOINT.

Example model: https://www.thingiverse.com/thing:4197735/files (choose "Wizard_Dice_5.4.stl").
OS: Windows 10
Browser: Chrome 88.0.4324.190 (a test in Firefox 86.0 is running in parallel but has not completed yet)

This is on a computer with a Core i5 CPU (8 threads), 16GB RAM, 512GB NVMe SSD

To compare with other slicers: Cura also crashes from similar models (but already after a couple minutes), Simplify3D, KISSlicer, PrusaSlicer finish slicing (with S3D bring fastest, KISSlicer taking a couple minutes, PrusaSlicer taking a longer time).

stewartoallen, 2021-02-26:
Thanks for submitting a torture test. As a rule of thumb, objects with more than 500k to 1M vertices will be a challenge. This one has 1.5M vertices. Settings such as "thin wall", infill type and %, and "min solid" will have an outsized influence on processing time.

Browsers, in general, do not allow more than 2 virtual CPUs to be used by any single browser tab. They also limit memory in ways that can seem arbitrary at times. Firefox's memory management is the most different from Chrome / Safari / Edge. In the past, I found it faster for small models and slower for large ones. But this could have changed.

I go through periods of performance refactoring. And I think that time is coming again. I will use this model to see how much faster and more memory efficient I can make Kiri:Moto.

Also on the todo list is conversion of some core routines to WASM. This can be faster in many cases. But marshalling (moving objects in and out of memory spaces between JS and WASM) can negate some of the benefit.

stewartoallen, 2021-02-26:
I was able to slice the model you provided in Edge on a 2014 MacBook Pro in just under 3 minutes with some default settings.

[image]

oliof, 2021-02-26:
Firefox 86 finished the slicing, but I didn't time how long it took. I did all slicing at 0.2mm layer, 0.35 infill on the default unmodified ender 3 profile.

oliof, 2021-02-26:
Oh yeah, this is on a 4 core/8 threads Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz. judging from Task Manager, Chrome only allows a single core only, while Firefox allows two.

oliof, 2021-02-26:
I can confirm Edge on Windows 10 behaves better and finishes the slicing in about 6 minutes on my machine.

oliof, 2021-02-26:
@stewartoallen did you test exporting the rendered file? I am 10 minutes in and the progress bar is at about a third of the way .... (with Edge).

stewartoallen, 2021-02-26:
I did not. Perhaps it hit a memory boundary and went into cleanup. The progress bar isn't as linear as it used to be. New code has thrown it off. It's on my TODO list to rescale portions of the calculation.

stewartoallen, 2021-02-26:
If you're on 2.6.3 and have `code preview` checked, I believe it fails to truncate the preview and consumes too much memory. 2.7 is better behaved in that regard.

oliof, 2021-02-26:
If `code preview` is checked by default, I have it checked. I am trying to not deviate too much from what I get out of the box. I seem to be on 2.6.3 ... is 2.7 available?

stewartoallen, 2021-02-26:
2.7 is the current development version. also live on the site. just go under the top-right user-profile menu and select version. it gets new code every 24 to 48 hours.

stewartoallen, 2021-02-26:
2.7 sliced, prepared, and downloaded the gcode in about 6 minutes in Edge on my 2014 MBP

oliof, 2021-02-27:
in 2.7 on Edge on my Lenovo Thinkpad E480, slicing not complete after 23 minutes ...

stewartoallen, 2021-02-27:
Is `thin walls` enabled under `layers`? if so, disable. it doesn't apply to this model. also, under preferences there is a `parts / quality` pull-down. the labels are misleading and i will change them, but set it to `poor` and it will go about 2x-3x faster. that's really a polygon "resolution" setting. for benchy, it should be `good`. but for highly textured models like this, it's overkill.

oliof, 2021-02-27:
thin walls is disabled (as is the default). I will check parts quality and try with poor. I will try on a Chromebook I have at my disposal (Pixelbook, so rather top of the line) for comparison tomorrow.