GridSpace/grid-apps #116: Feature Request: Add support for uploading gcode directly to RepRapFirmware

oliof, 2021-02-27:
Most RepRapFirmware setups are network connected, so it would be nice to be able to upload (and start printing of) gcode files directly from Kiri:Moto. Refer to the [Developer Documentation](https://raw.githubusercontent.com/wiki/Duet3D/RepRapFirmware/HTTP-requests.md) for the REST API.

An example implementation of this functionality in python can be seen at https://github.com/Kriechi/Cura-DuetRRFPlugin

stewartoallen, 2021-02-27:
I recently acquired a Duet 2 Wifi for testing exactly this. There are some difficulties, however.

First, DWC needs to present an https endpoint. Browsers no longer allow mixed secure / non-secure.

Second, DWC needs to allow certificates (self-signed ok).

Third, the DWC endpoint needs to be mapped to DNS that matches the ssl certificate.

Last, the DWC endpoint needs to enable CORS headers.

My stop-gap solution was to add a "sender" endpoint to [Grid:Host](https://github.com/GridSpace/grid-host) as discussed in [this thread](https://forum.grid.space/t/grid-host-how-to-install/265/10).

Grid:Host is an old project of mine that allows Kiri:Moto to spool directly to proprietary end-points or devices on your local network. Subsequently, I have added the notion of GridLocal devices which bounce a print stream off of a secure cloud endpoint. But that will not work in this case because I cannot run custom server code on the DWC.

oliof, 2021-03-01:
The Reprapfirmware documentation indicates enabling TLS should be possible (see https://duet3d.dozuki.com/Wiki/Gcode#Section_M586_Configure_network_protocols). I have not tried that myself yet.

oliof, 2021-03-01:
I spoke to soon:

> Note: TLS has not yet been implemented in RepRapFirmware, therefore T1 will not work.

stewartoallen, 2021-03-01:
I read up on that. As soon as you depend on an external PI to run code, you might as well run the existing GridHost code which already runs on a PI and will provide spooling from KM to the DWC.