GridSpace/grid-apps #122: Arc moves not compatible with RRF 2.x

oliof, 2021-03-03:
Hi,
when simulating a GCode file generated by kiri:moto 2.7D14
[MW_A-011-gcode.txt](https://github.com/GridSpace/grid-apps/files/6074600/MW_A-011-gcode.txt)
 with arc set to 0.15, RepRapFirmwaere 2.0.5.1 (last release in the 2.x series) runs into errors:

```
Error: SetPositions called when DDA ring not empty Cancelled simulating file 0:/gcodes/MW_A-011.gcode after 0h 3m simulated time
Warning: Tool 0 was not driven because its heater temperatures were not high enough or it has a heater fault
Error: G2/G3: bad combination of parameter values
```

On RepRapFirmware 3.3b1 (current preview release of the 3.x series), the simulation completes without error.

GCode attached.

I will also post this to the duet forums and update with what they think.

oliof, 2021-03-03:
[MW_A-011-gcode.txt](https://github.com/GridSpace/grid-apps/files/6074605/MW_A-011-gcode.txt)

oliof, 2021-03-03:
Accompanying thread on the duet forums: https://forum.duet3d.com/topic/21898/kiri-moto-arc-moves-incompatible-with-rrf2-x-work-in-3-x

oliof, 2021-03-03:
Confirmed that this is likely a bug in RRF that is (finally) fixed in RRF3.x. So nothing to do here, move along (-:

oliof, 2021-03-03:
Someone with more patience than me found two spurious lines in the generated gcode:

> I can see line 2968 might be a problem.
> (All the g3s are NaN for the R parameter)
> 
> G3 X-18.9677 Y9.8100 RNaN E0.6945
> 
> [HERE'S](https://ncviewer.com) my favorite GCODE viewer that does G2/3 right.
> This might also be a problem.
> 
> G2 X12.7817 Y-7.4774 R21230140343422.7930 E0.0855 ; merged=2 len=3.2712 cp=15011976202389.71,15011976202366.37

stewartoallen, 2021-03-04:
clearly wrong. can you share a workspace from which I can replicate the creation of this gcode?

oliof, 2021-03-04:
Workspace: 
[workspace-122.zip](https://github.com/GridSpace/grid-apps/files/6086976/workspace-122.zip)
Output with an RNaN from that workspace:  
[MW_A-015.zip](https://github.com/GridSpace/grid-apps/files/6086981/MW_A-015.zip)
Source STLs: https://www.thingiverse.com/thing:4545473