GridSpace/grid-apps #125: Adaptive Toolpaths

Harvie, 2021-03-04:
I will just leave this here as a sign of wishful thinking :-)

https://knowledge.autodesk.com/support/inventor-cam/learn-explore/caas/CloudHelp/cloudhelp/2020/ENU/INVCAM/files/GUID-970298C7-2F2E-47E6-AD54-454DEC0BBD7C-htm.html

[image]

[image]

stewartoallen, 2021-03-04:
Looks beautiful. It's already somewhere on my TODO list. Can't wait to figure out the algorithm for this.

Harvie, 2021-03-04:
I can offer you my collection of links on this topic i've gathered so far:

https://www.freesteel.co.uk/images/blog/brochure.pdf
https://github.com/jf---/libactp
https://github.com/Heeks/libactp-old
https://github.com/kreso-t/FreeCAD_Mod_Adaptive_Path
https://forum.freecadweb.org/viewtopic.php?f=15&t=30127
https://github.com/vlachoudis/bCNC/issues/815