GridSpace/grid-apps #126: Model vanishes after slicing, no error

oliof, 2021-03-04:
A model I was previously able to slice now seems to vanish from the plater after slicing. It's a simple fan duct, nothing fancy, and I printed it successfully with the original slice -- only needed to increase temperature and layer height a bit so I tried to slice it again.

Unfortunately, I did not save the original workspace. The current one is attached here. 

Original model [EVA](https://main.eva-3d.page/drives/hemera/stls/Duct%20v2.1%20Angled%2013.2deg.stl)
Workspace 
[workspace-vanish.zip](https://github.com/GridSpace/grid-apps/files/6087154/workspace-vanish.zip)

Generated G-Code when hitting export is just the machine's start and end gcode. Thin walls is enabled.

stewartoallen, 2021-03-04:
Interesting. No errors on the Javascript console? Seems to be slicing for me when I import your workspace.

[image]

oliof, 2021-03-05:
Reslicing this morning in 2.7D16, itworks. The kiri:moto window was open for a longer time, maybe some kind of memory leak?)

stewartoallen, 2021-03-05:
it doesn't use memory or do any work unless it's actively slicing or previewing. in future, if anything odd happens, just look on the javascript console first.

stewartoallen, 2021-03-05:
re-open this ticket if you are able to repeat the issue

closed 2021-03-05 by stewartoallen

oliof, 2021-03-08:
Hi,
I just had this happen again. JavaScript Console shows no errors other than

`THREE.Quarternion: .inverse() has been renamed to invert().`

at the begining of the session.

I can't attach the workspace file because even zip compressed it's larger than 10MB. You can download it from [Google Drive](https://drive.google.com/file/d/11VQAGfhIvh5cBlXcxpMkhS2vugARcsgU/view?usp=sharing).

stewartoallen, 2021-03-08:
I got an out of memory crash with that model. Perhaps that is the cause for you. I don't get the disappearing act, though.

That THREEjs warning is part of their library. I can't disable it. Hoping some new version from them will remove it.