GridSpace/grid-apps #127: FR: draft shield

oliof, 2021-03-05:
Draft shields are like skirts, only they extend for the full height of the model (alternately, a specified height). They can enable warp-free printing of PETG and ABS models on printers that do not have enclosures by protecting a print from moving air (drafts) and trap the heat a bit.

stewartoallen, 2021-03-16:
Currently rafts and skirts / brims are generated in path routing. This is due to slicing being historically handled on a "per widget" basis. And these structures can span multiple objects. A bit refactoring must take place to lift rafts + skirts + brims into the slicing process before draft shields can be cleanly implemented. This also applies to #128

stewartoallen, 2022-10-11:
this was implemented in 3.4 or 3.5

closed 2022-10-11 by stewartoallen