GridSpace/grid-apps #129: electron version?

0xtrzy, 2021-03-05:
Is possible create electron version for linux?

stewartoallen, 2021-03-05:
Yes, I have not looked into this in quite some time. I would be happy to accept a code contribution if someone did the work.

ConorIA, 2021-07-27:
Not sure if this is of interest, but there is a PoC for a QT wrapper for a different webapp over at https://gitlab.com/ddobrev/Fibre. It could be adapted for Kiri:moto in a manner of minutes. It is lighter than electron, but I couldn't provide a full pro/con list for one or the other.

pbek, 2021-07-30:
Doing something with https://nwjs.io/ also shouldn't be too hard.

pbek, 2021-07-30:
> Not sure if this is of interest, but there is a PoC for a QT wrapper for a different webapp over at https://gitlab.com/ddobrev/Fibre. It could be adapted for Kiri:moto in a manner of minutes. It is lighter than electron, but I couldn't provide a full pro/con list for one or the other.

I'm not sure Qt webengine supports WebAssembly...

pbek, 2021-07-30:
I gave it a few hours, see https://github.com/GridSpace/grid-apps/pull/181. You can already run the app on a desktop with `npm run start`. Needs more work.

pbek, 2021-07-31:
But it looks already nice...

[image]

pbek, 2022-06-10:
The PR never got merged. 😅

stewartoallen, 2022-06-20:
at this point the PR conflicts. also, it really gets into the package file -- is there a way to create a more stand-alone package or put it all into a sub-directory?

sp00ck, 2022-06-28:
The internet will not always work and not everywhere. I just want to be able to work completely offline. Without access to the Internet. At home on a disk. 

I do not care in what form. Of course it would be best if the electron contained everything as in the 'large' installation to make it easier to maintain. 
But for a user the most important thing would be one file without problems with dependencies, libraries etc. That's why electron and similar were created. A directory is not the ideal solution when it comes to dependencies.

less problem == electron (or docker but it is not this same easy way and not on all platforms)

pbek, 2022-06-28:
#181 was based on https://nwjs.io/, feel free to take over where I stared.