GridSpace/grid-apps #133: External thin walls are treated like infill

oliof, 2021-03-09:
Posting by request: External thin walls are treated like infill and have the wrong speed applied (standard print speed, rather than finish):

[image]
[image]

In practice this will rarely be an issue due to acceleration and jerk limiting max speeds on these, but it should be fixed nonetheless.

stewartoallen, 2021-03-12:
This is fixed in the experimental thin-fill in the 2.8 development version. Just check experimental under preferences.

stewartoallen, 2022-10-14:
re-open if this is still an issue. the thin wall code is being completely re-written in the next release.

closed 2022-10-14 by stewartoallen