GridSpace/grid-apps #137: When skirt distance is zero (i.e., when using a brim), always print outside in.

oliof, 2021-03-14:
When skirt distance is zero (or close to zero), I want a brim that helps with adhesion. But for better overhang performance, I usually print walls inside out. I also try to avoid a prime/purge line when using skirt/brim.

With brim and initial layer walls printing inside-out, I get:

a) an imperfect brim if the nozzle isn't primed.
b) an imperfect mating of the walls and brims because both brim and wall are printed inside out.

Brims and walls attached to brims should always be printed outside-in to allow for the adhesion help and a clean connection.

current workaround: Select layer 0 as a range, and set outside-in for the range. Maybe this is sufficient, then feel free to close this bug. But kiri:moto does "the right thing" automagically in many places, so maybe this should be one too.

oliof, 2021-03-14:
I did test the ranges workaround and it looks like brims are always printed inside-out. I think they should either follow the wall setting _or_ always print outside-in.

stewartoallen, 2021-03-14:
the advantage of inside-out is that the nozzle crosses the brim when traveling back to start the part hopefully clearing any oozing from the nozzle. but I will make this change for when the brim is close or attached to the part.

commit 8e7a51a0 (2021-03-14): fixes #137
closed 2021-03-14 by stewartoallen