GridSpace/grid-apps #141: Cannot GET /kiri" error

enzoFR60, 2021-03-15:
Hello, I can't access http://localhost:8080/kiri (Cannot GET /kiri)

I looked at the readme, I reinstalled several times, but I can't access it ...

Please help me

stewartoallen, 2021-03-15:
if you add a trailing `/` does it work? `http://localhost:8080/kiri/`

enzoFR60, 2021-03-15:
No it doesn't work,

Cannot GET /kiri/

logs:
210315-015649 "[appserver]" "app-server running: ports=[8080] apps=[apps] data=[data] logs=[logs]" 
210315-015651 ["GET","localhost:8080","/kiri/","::1","","Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.72 Safari/537.36"]

stewartoallen, 2021-03-15:
in this setup

```git clone git@github.com:GridSpace/grid-apps.git
cd grid-apps
npm i
npm install -g @gridspace/app-server
gs-app-server --debug
```

do you miss the `cd grid-apps` step?  if the output doesn't look like this, you may have:

```
210315.091712 '[appserver]' 'initialized grid from apps/grid'
210315.091712 '[appserver]' 'app-server running: ports=[8080] apps=[apps] data=[data] logs=[logs]'
```

it should say `initialized grid from ...` on the first line

stewartoallen, 2021-03-15:
also, if you are running Windows, note the `Windows Developers` caveat below on the same setup page.

enzoFR60, 2021-03-15:
Yes I have checked but I do not have the "'initialized grid from apps / grid', I will test on a debian

enzoFR60, 2021-03-15:
actually the problem comes from windows

closed 2021-03-15 by enzoFR60