GridSpace/grid-apps #142: it is possible to use it as an API in post or in get?

enzoFR60, 2021-03-15:
Hello,

it is possible to use it as an API in post or in get, to retrieve the printing time, fillament size, ...?

If so, an idea how to do it, because I don't know how to work with this style of library

thank you in advance

stewartoallen, 2021-03-15:
Have you looked at the various APIs? https://docs.grid.space/projects/kiri-moto/apis

enzoFR60, 2021-03-15:
@stewartoallen 
Yes I looked but not all.

@AlbertMN
Okay, I'll try to play with it all,
I explain my use,
it's for a 3d printing website, where we load a stl, and I would like to have "filament used (mm)" to estimate the price, print time.

thank you for your answers

stewartoallen, 2021-03-15:
I think this is what you're looking for: https://grid.space/kiri/engine.html

kinj14, 2021-03-15:
I saw this, but I have trouble seeing how to turn it into an api

stewartoallen, 2021-03-15:
if you want to estimate price and print time accurately, you should do it from the gcode that will be used to make the final product.

kinj14, 2021-03-15:
I know, I use "grid-apps" for this,
The goal is that the customer upload a .stl, .obj and that afterwards we print it but for that I need a price or an estimate

kinj14, 2021-03-15:
I just have an idea, call the page that retrieve a link by the url, and load and slice the model, which modifies the page with the information

I will try this

stewartoallen, 2021-03-15:
You should be able to include the engine as a script in your page and do it all within the page.

kinj14, 2021-03-15:
On a basic web server (example: apache/php) ?
Which file to take?

stewartoallen, 2021-03-15:
the script goes into the page html. it's not a server-side thing. the example shows this.

kinj14, 2021-03-15:
yes but i tried using it on my page but it didn't work

stewartoallen, 2021-03-15:
can you be more specific?

stewartoallen, 2021-03-15:
I recommend hopping on the Grid.Space Discord server and going to the `#developers` channel

closed 2021-03-16 by stewartoallen