GridSpace/grid-apps #151: FR: advanced cooling settings

oliof, 2021-03-28:
Since cooling adversely affects interlayer adhesion, it would be good to allow for _some_ cooling, without cooling all of the print all of the time. Thus, this feature request is about adding the following options:

- cooling for overhangs and bridges
- cooling for the outermost line/layer (skin/shell/top)

Ideally, it should be possible to disable cooling for infill and inner walls. Even more ideally, different cooling settings for overhangs and outermost line/layer than just on/off would be possible.

stewartoallen, 2021-03-28:
Have you seen this work in practice? Since most fans don't spin up instantly, I wonder if the lag would have an effect. And if cycling the fan would shorten its lifespan.

oliof, 2021-06-11:
Yes, I've seen this work in practice. Recent versions of superslicer have an option to set fan startup delay so the fan commands are injected early enough so the fan runs at desired speed when the time for that has come.

filipgoc, 2022-02-21:
Yes, this is a pretty standard feature in most slicers these days. Bridging does deserve its own settings in general. For example, some materials have lower layer adhesion and strength when cooled excessively so I'll be running the fan as low as possible, but for bridges I still want to go high to have better bridges.

stewartoallen, 2022-02-21:
what's interesting about cranking fans on bridges is that it's contra-indicated by some tests. like this:

https://twitter.com/repkord/status/1482167496132030464

was printed upside-down and still had a "dip" on the middle of the bridge. that could have only been caused by the fans pushing on it since gravity was in the opposite direction. so for bridging, I would expect 1) lower fans and 2) slightly under-extrude to pull the line tight.

having said all that, separate bridge settings are on the todo list. will be in one of the next two point releases.