GridSpace/grid-apps #152: Different paths in "slice", compared to "preview"/"export" (possibly thin lines related)

oliof, 2021-03-29:
In the attached workspace I get thin lines in the "slice" view of an object, but they do not turn up in "preview" or "export".
This is obviously unexpected and undesired.

[thin_not_in_export.zip](https://github.com/GridSpace/grid-apps/files/6222002/thin_not_in_export.zip)

oliof, 2021-03-29:
Compare the following images which were sliced with no difference in settings:

[image]
[image]

The first one is what I'd expect, the second isn't.

stewartoallen, 2021-03-30:
I see the confusion. The engine has identified that area as a gap (red), but has not filled it. The red is not an extrusion line. It's a debugging line. I added it when debugging the experimental thin-fill, which you have enabled here.

[image]

stewartoallen, 2021-03-30:
I'm going to close this because it's not really a bug or a feature request. I do have on my todo list to resolve the multiple versions of thin-wall or just add a selector for it.

closed 2021-03-30 by stewartoallen