GridSpace/grid-apps #158: outline never starts from the top

marciocoragem, 2021-04-25:
The outline never starts from the top on some shapes.
[image]

I have tried multiple combinations and nothing works.

stewartoallen, 2021-04-25:
Outline never starts at the top of a part since it is meant to remove material. You can force tops and faces to be cleared with a roughing operation or you can use a trace/clear operation to clean one face of a part.

marciocoragem, 2021-04-25:
I understand that but as you can see in the imagem it starts a lot lower than supposed to, and why in the holes it starts from the top?
In this file i'm using a rough with 1mm depth of pass and them use a outline with 0,2 to clean up the surface (kind of like a finishing pass with a ballnose).

This works nice and gives a good finishing surface everywhere except in the first 2 or 3mm.

marciocoragem, 2021-04-25:
Another thing is that i used the contour feature before with success, but now i have some problems. I have select the options use only curves and inside only, but it generates toolpaths in the top of the stock, like you can see in the images.
[image]

stewartoallen, 2021-04-26:
Please right-click and export your workspace. Then either drop a zipped up `.km` file here or email it to sa@grid.space -- thank you

marciocoragem, 2021-04-26:
Thanks for taking a look.

[workspace.km.zip](https://github.com/GridSpace/grid-apps/files/6375233/workspace.km.zip)

stewartoallen, 2021-04-26:
I have a fix for the curves only contouring that I will push out later today.

Before

[image]

After

[image]

stewartoallen, 2021-04-26:
Your other problem is caused by the geometry of the part. The interior pocket is joined at the top by two exterior pockets that extend to the outside of the part which causes them to be considered "outside". So the outline inside-only option you have checked causes those cuts to be removed.

[image]

With the way the outline operation works at present, the workaround is this:

disable `inside only`
set stock offset to 0, 0, 0
only mill one part at a time so the stock clipping will work as expected

[image]

marciocoragem, 2021-04-26:
Thanks,
I will try those settings.

This is by far my favorite free CAM software, i really like how it works with stl files.

closed 2021-04-26 by stewartoallen