GridSpace/grid-apps #159: Contour tool path optimization [CNC]

geobruce, 2021-05-02:
Tool path optimization. (traveling sales man problem)
[image]
In the middle of the ring is no work done while it would be a lot more efficient to just do a step to the side and mill that curve.
[image]

And doing an offset of the outline or circles or spirals might be very efficient as well just like in [another enhancement](https://github.com/GridSpace/grid-apps/issues/150).

stewartoallen, 2021-05-02:
the `curves only` option will eliminate the inner cuts. but I agree. other strategies would be very useful.

stewartoallen, 2022-10-10:
a bridging option has been added to contouring to address this. also check out the latest `contour` option in `pocketing`.