GridSpace/grid-apps #160: Objects Sliced in Air

Waldkaeuzchen, 2021-05-03:
Hello,
I have got that problem that multiple Parts are shifted into the Air after the slicing-Progess, making the resulting G-Code useless.

Preview:
[image]

Result after Slicing:
[image]

Single Part Sliced:
[image]

G-Code and Profile:
[Files.zip](https://github.com/GridSpace/grid-apps/files/6415329/Files.zip)

stewartoallen, 2021-05-03:
This has been fixed in 2.9 -- use the top-right person/gear icon to switch versions

Waldkaeuzchen, 2021-05-03:
Oh, thank you.

closed 2021-05-03 by Waldkaeuzchen