GridSpace/grid-apps #161: 3+2 (positional 5 axis) CAM

smartin015, 2021-05-04:
I've been exploring Kiri:Moto for a few days now - it's been excellent for sending prints to my CR30 belt 3D printer where most other slicers don't really have a well lit path.

I'd love to also use this with my [PocketNC](https://pocketnc.com/), which is a 5-axis desktop CNC mill. Full 5-axis machining would be cool, but honestly I find myself operating in 3+2 most of the time by rotating AB to a certain orientation, then proceeding with 3-axis style cuts. 

As I understand it, Kiri:Moto gets halfway to this point already with the Flip operation changing the orientation of the part. If that supported e.g. clicking a face to set the XY plane, it'd allow for chaining different axis setups without a bunch of manual gcode stitching. I'd be happy to provide gcode samples that have worked for bed rotation on the PocketNC - I also don't have a lot of experience with its [postprocessor](https://pocketnc.com/pages/post-processor) but I'll do what I can to support this feature.

The most cost-effective solution I've found that does this today is Fusion 360, but it's a special feature which requires the $60/mo membership which is a persistent drain on anyone thinking about multi-axis machining in a home shop. With a 3+2 workflow, Kiri:Moto would basically be the only free and open-source solution around that could do it. Please consider - thanks!

smartin015, 2021-05-04:
Thanks for the quick reply and your thoughts! Yes, that's basically what I'm looking for. Exciting it fits into your existing plans :)

Re: awareness of prior steps, I haven't found a way to do this even in F360, short of manually adding planes/constraints which effectively copy over the volume removed in a previous operation. Usually that means my workflow looks something akin to a flip jig setup - with a clearing pass, a second one at 180 degrees, then a few additional off-axis passes to bore holes, clean up certain faces, etc. If the slicer is aware of space that's already been cleared then it could open up some pretty neat opportunities.

smartin015, 2021-05-15:
Found some time to try out kiri:moto with the PocketNC - got to make some brass chips! Here's the workspace & machine setup I used (warning: machine setup assumes only 1 tool and traditional 3-axis without rotations)

[pocketnc.zip](https://github.com/GridSpace/grid-apps/files/6487722/pocketnc.zip)

And just for fun, here's a pic of where I ended up. I wasn't able to complete the run because of issues with tool pull-out from the collet (I have a support ticket in progress about that).

[image]

Possibly helpful to confirm behavior for 3+2, there's also a [web simulator](https://sim.pocketnc.com/). I use it every time I export a new file as a safety check before running it on the real machine.

stewartoallen, 2021-05-17:
Awesome :) thanks for posting this

legontanon, 2021-06-30:
Meshcam is $250...
http://www.grzsoftware.com/

legontanon, 2021-06-30:
I think the task is way more complicated than just changing planes and 2d slicing...

- The toolpaths must be spiral to work at decent speeds and be able to reach cavities 

- Then to do things well you should be cutting perpendicular to each triangle the mesh
   - remesh to the desired precision
   - find the paths in the mesh thru adjacencies

At this point you have a toolpath  whose XYZABC are relative to the piece...

After that you have the inverse kinematics to deal with... 
Each machine has it's own... fundamentally you have to convert each position in the part to the actual (absolute) position of the motors...

That's not really a task for the the toolpath generator, that should be a gcode postprocessor that converts part relative movements  (XYZABC) into absolute ones (XYZAC in your case). That is done just mutipliying the relative vectors by few matrices 
(check this one http://www.ijmerr.com/uploadfile/2020/0109/20200109113341152.pdf)

I'm planning to get into it (the preprocessor with inverse kinematics) soon still meditating which existing OSS preprocessor to start from

 and later I might go for the toolpath generation, I'm just investigating whether Kiri:Moto is a good starting point for that...

smartin015, 2021-07-29:
MeshCAM doesn't actually do 3+2 machining, only 2-side (i.e. flip jig) machining, so it's not really what I'm looking for. I wish it was though... $250 for a permanent license would've been great.

I'd be satisfied with even a proof of concept at this point. I'm not asking for true 5ax machining with all the optimizations, just the ability to pick a perspective and treat it as 3-axis like KM already does, but with some GCODE I can specify to glue each oriented operation together into a single gcode file. I can already do this manually (rotate the part, export, rotate again, export etc.) but that's a lot of effort and is limited by the number of orientations I'd actually want to do by hand.

@stewartoallen I find myself with some extra free time in the next month or so. If I wanted to try implementing this, could you advise? At first glance I think I'd need to:

1. Add a new `OpRotate` or some such to mode/cam/ops.js 
2. Update the `op-add-list` logic [here](https://github.com/GridSpace/grid-apps/blob/e19c27170aff2f4bc96b3f19ff106824fe08d4ed/src/mode/cam/client.js#L155) to include a "rotate" operation
3. Edit the [export flow](https://github.com/GridSpace/grid-apps/blob/master/src/mode/cam/export.js#L254) to remap points based on applied transformations for each operation

Does that make sense to you / fit with your roadmap, or am I missing something?

stewartoallen, 2021-07-29:
This is directionally where I was headed with 4th axis support. The hardest part of implementing this is doing justice to some sort of preview and/or animation. Also, preserving and restoring part orientation so you're not "messing up" the workspace by playing with parameters.

smartin015, 2022-02-12:
Just wanted to give a little update after all these months - I'm still working on an open-source solution for 5-axis CAM, albeit not directly in KM. I've published what I have so far in https://github.com/smartin015/5axis_cad, but it's still in very early stages and not intended for other folks to just pick up and use.

It's developed on top of [cadquery](https://cadquery.readthedocs.io/en/latest/) and lets you specify a sequence of subtractive operations from a stock volume, mostly focused on radial operations since that's what I find interesting at the moment. Side benefit of this approach: the operations used to construct the 3D model are the same operations that cut it out. 

There may still be some benefit to KM insofar as I'm writing GCODE generator functions, but it's in a different language, with a different spec, and following a different paradigm, so the overlap may not actually be that much.

stewartoallen, 2023-08-31:
I'm on an extended break from coding work to focus on other projects. Still monitoring the forums and discord, though.