GridSpace/grid-apps #164: outline operator not doing anything

geobruce, 2021-05-13:
I tested this in version 2.8 up to 3.0 dev.
[image]
[image]

Am I doing something wrong?

[outline opperation broken.zip](https://github.com/GridSpace/grid-apps/files/6472438/outline.opperation.broken.zip)

stewartoallen, 2021-05-13:
a bunch of check boxes are missing from your outline operation. and I'm trying to figure out how it got that way.

[image]

stewartoallen, 2021-05-13:
open up your javascript console and paste in these two lines to fix it:

```
kiri.api.conf.get().process.ops[0].inside = false
kiri.api.conf.get().process.ops[0].outside = false
```

stewartoallen, 2021-05-13:
if you can re-create the steps to get back into this state, please document them so I can fix it.

geobruce, 2021-05-15:
this doesn't seem to fix it
[image]
Still the same check boxes

stewartoallen, 2021-05-15:
if you open Kiri:Moto in an incognito window, does the outline operation show all the checkboxes as pictured above?

geobruce, 2021-05-15:
Only 4 of the 5 shown in ver: 3.0 Dev 
[image]

stewartoallen, 2021-05-15:
uncheck `outside only` ... the list is adaptive and hides mutually exclusive options. you can't, for example, choose both `inside only` and `outside only`

geobruce, 2021-05-15:
oops that way it does work in incognito maybe I should just clear cache, cookies, etc..

stewartoallen, 2021-05-15:
i'm really not clear on how your outline op got into that state. i need a repeatable set of steps. the commands I gave you should have cleared up any mutually exclusive settings *if* the only operation in your chain was the `outline` op.

stewartoallen, 2022-10-10:
if this is still an issue, please export your workspace (right click export) and zip it so you can attach it into this report.

closed 2022-10-10 by stewartoallen