GridSpace/grid-apps #165: Unexpected contour behavior

geobruce, 2021-05-13:
All vertical walls have straight edges except for the inner "dome". 
[image]

[image]

But I see a lot of tool paths around the straight edges.

[straight edge countour weird behavior.zip](https://github.com/GridSpace/grid-apps/files/6472479/straight.edge.countour.weird.behavior.zip)

stewartoallen, 2021-05-13:
because you have a ball mill selected, it generates curved paths over edges

[image]

same settings with an standard endmill

[image]

stewartoallen, 2021-05-13:
the algorithm as it stands looks for linear horizontal cutting paths and removes them. it will have to be re-imagined to exhibit the behavior you're expecting. but now that I see it, I can think of ways to achieve it. so I will add it to the 3.0 list.

closed 2022-10-10 by stewartoallen