GridSpace/grid-apps #175: leveling cnc

zerotrzy, 2021-07-01:
Why leveling is on whole space? why not only over the object?
In my opinion add a checker [x] only over object

[image]

stewartoallen, 2021-07-01:
`clear top` under the roughing operation does what you want. the level op is designed to create a flat surface over the entire stock.

[image]

closed 2021-07-01 by stewartoallen

ojej, 2021-07-02:
flat surface not working if my flute is small, I get shape not flat area