GridSpace/grid-apps #179: FR: Command Line Interface (CLI)

Spiritdude, 2021-07-26:
I like to use the slicer on the command line (without the browser), something like 
```
% KiriMotoSlicer cube.stl -o cube.gcode
```
I have a model -> slice -> gcode pipeline supporting multiple slicers (slic3r, prusa-slicer, cura, etc) and I like to use KiriMoto as well as alternative.

Is anyone working on it currently or has it been done (and I didn't find it)?

What would be the best way:
- create pure NodeJS version from browser JS code (I tried, but I ran quickly into browser-JS vs Node-JS module incompatibilities)
- extend app-server to slice and execute with STL file as single argument (+ config), slice and exit again
- create a small wrapper sending STL to (locally running) app-server (which slices too) and get .gcode back and save it as a file - thereby create server API and scalable slicing framework

stewartoallen, 2021-07-26:
There is a way to run the slicer from the command line using node and without requiring the app server. It requires wrapping some of the dependent libs to fake a browser environment. I will check in an example soon.

stewartoallen, 2021-08-02:
sample that can be expanded https://github.com/GridSpace/grid-apps/commit/cb17b786689c9bbb9f889eee7af7980030da95c1

closed 2021-08-02 by stewartoallen

Spiritdude, 2021-08-02:
Thanks for the code, yet, when I tried it:
```
% nodejs --version
v10.19.0

% nodejs src/kiri/cli.js
{ dir: '.', file: './src/geo/debug.js' }
{ fake_worker: '/code/worker.js?undefined' }
kiri | init work | rogue
{ fake_fetch: '/web/obj/cube.stl' }
{ loaded:
   Engine {
     widget:
      Widget {
        id: 'kru7605i0',
        group: [Array],
        roto: [],
        adds: [],
        mesh: [kn],
        points: null,
        bounds: null,
        wire: null,
        slices: null,
        settings: null,
        modified: true,
        track: [Object],
        cache: {},
        stats: [Object],
        meta: [Object],
        support: false },
     settings:
      { mode: 'FDM',
        controller: {},
        render: false,
        filter: [Object],
        device: [Object],
        process: [Object],
        widget: [Object] },
     listener: [Function] } }

```

it never returns or exits.

Is it a functional example or is it a non-functional skeleton which needs further development?

stewartoallen, 2021-08-02:
I just tested it and it's working for me. My node version is `v14.15.5` but that shouldn't make a difference.

Spiritdude, 2021-08-02:
Could you post the terminal output of yours?

I started to insert debug infos:
```
   //.then(() => something()) 
   .then(function() { console.log("step 1") something() })
```
and 
```
    //.then(eng => eng.slice())
    .then(function(eng) { 
       console.log("eng.slice()") 
       eng.slice() 
    })
```
and finally get an error message (first 4 lines are debug outputs):
```
moveTo(1,1,1)
setProcess()
setDevice()
eng.slice()
{ error:
   TypeError: Cannot read property 'slice' of undefined
       at /home/spirit/Projects/KiriMotoSlicer/grid-apps/src/kiri/cli2.js:318:12
       at process._tickCallback (internal/process/next_tick.js:68:7)
       at Function.Module.runMain (internal/modules/cjs/loader.js:834:11)
       at startup (internal/bootstrap/node.js:283:19)
       at bootstrapNodeJSCore (internal/bootstrap/node.js:623:3) }

```

stewartoallen, 2021-08-02:
```{ dir: '.', file: './src/geo/debug.js' }
{ fake_worker: '/code/worker.js?undefined' }
kiri | init work | rogue
{ fake_fetch: '/web/obj/cube.stl' }
{
  loaded: Engine {
    widget: Widget {
      id: 'krurpfpd0',
      group: [Array],
      roto: [],
      adds: [],
      mesh: [kn],
      points: null,
      bounds: null,
      wire: null,
      slices: null,
      settings: null,
      modified: true,
      track: [Object],
      cache: {},
      stats: [Object],
      meta: [Object],
      support: false
    },
    settings: {
      mode: 'FDM',
      controller: {},
      render: false,
      filter: [Object],
      device: [Object],
      process: [Object],
      widget: [Object]
    },
    listener: [Function (anonymous)]
  }
}
{
  gcode: '; Generated by Kiri:Moto undefined\n' +
    '; Mon Aug 02 2021 11:08:48 GMT-0400 (Eastern Daylight Time)\n' +
    '; Bed left:0 right:220 top:220 bottom:0\n' +
    '; Bed type: fixed\n' +
    '; Target: internal\n' +
    '; --- process ---\n' +
    '; processName = default\n' +
    '; sliceHeight = 0.25\n' +
    ...
```

stewartoallen, 2021-08-02:
can you try a newer version of node and see if that helps?

Spiritdude, 2021-08-02:
I used `nvm` to install `node 14.15.5` as you have, and it **worked**!

Spiritdude, 2021-08-02:
Thanks a lot for `cli.js`, I will start to extend.

Spiritdude, 2021-08-02:
node `10.0.0` -> stalls (never exits)
node `11.0.0` -> core dumps
node `12.0.0` -> works
node `13.0.1` -> works
node `14.0.0` -> works