GridSpace/grid-apps #182: Algebra on temp token fails

oliof, 2021-07-31:
In both 2.9.5 and 3.0.D15, algebra on temp token doesn't work. I have the following as a makeshift temperature offset for second tool on my IDEX system in the start gcode

```
[...]
T1              ; select second tool
M104 {temp+10}  ; set extruder temperature
[...]
```

And for a filament temperature of 215 per profile I get

```
[...]
T1
M104 21510
[...]
```

in the generated GCode file.

stewartoallen, 2021-07-31:
this means `temp` is being stored as a string. will find and fix. thanks

stewartoallen, 2021-07-31:
fix will be live tonight. shouldn't the expression be `M104 S{temp+10}`

oliof, 2021-07-31:
yes, it was a faulty reproduction. The fault still is true.

stewartoallen, 2021-08-01:
fix pushed to production. a workaround could have also been:

```M104 S{parseInt(temp)+10}```

commit 3a042eb7 (2021-07-31): fixes #182 numbers in expressions
closed 2021-08-01 by stewartoallen