GridSpace/grid-apps #184: Plunge rate discrepancy?

Stoli123, 2021-08-05:
The plunge rate for the first Z-axis change in each pass does not comply with the setting. 

Behavior: I I set up a simple project with 3 holes in a piece of stock, each hole to be 4mm deep, and set the step down depth to 1mm. So the resulting g-code will make 4 passes to complete the carve.  When I start the carve on my cnc, the initial plunge rate for the first hole is much faster than the rate which I set in Kiri:Moto. For the subsequent 2 holes in that pass, the plunge rate is correct. When the 3rd hole is cut and the next pass is started , the initial plunge (from 1mm to 2mm) is very fast again. For the remained of this pass, the rate is correct. This repeats with each pass.

stewartoallen, 2021-08-05:
please right-click export your workspace and email [sa@grid.space] or DM it to me on the Discord server. thanks

stewartoallen, 2021-08-06:
The plunge rate for the drill op was set to 300, which matches the generated gcode. There is also a feed limit setting under the limits menu, which covers any Z movement which is not explicitly drlling.

[image] [image]

stewartoallen, 2022-10-09:
closing this since it appears the gcode matched the workspace settings. If you have another example of this, please attach to the ticket (as a zipped workspace), and I can re-open it.

closed 2022-10-09 by stewartoallen