GridSpace/grid-apps #185: [Feature request](CNC) Waterline toolpath

geobruce, 2021-08-11:
Hey
I'm looking for a way to have a waterline like finish more efficient. Kind like the finish in the picture.
[image].
More specific in the red circle 
[image]
I know I can get this by doing a rough operation. But I'd like to make a real big landscape and with a water line resolution (step down) of lets say 1mm. But to clear all the unneeded material 1mm would be way to slow. I could easily clean this up with a 6mm step down.

So what I'm looking for is a workflow that looks like this:

1. Rough away all unneeded material with a step down of 6mm, (leave stock 1mm)
2. Do a waterline operation with a step down of 1mm which just passes the only material that's left. and not all the empty space again. (this is kind of "rest machining" related" but maybe it can be done easier by adapting the rough operation to a waterline operation. (if there is no toolpath under this toolpath perform the toolpath, If this is not the lowest toolpath Z-wise skip this one.)

Thanks in advance!

stewartoallen, 2021-08-14:
This is what the `outline` operation is for. Use `leave stock` in the `rough` operation then `outline` is the final pass.

geobruce, 2021-08-19:
Hello Allen

Today I had finally time to test this.

But with inside only I only get this:
[image]
[image]

When I disable inside only I get this:
[image]
[image]
Here is the workspace if you'd like to try it for your self: [workspace.zip](https://github.com/GridSpace/grid-apps/files/7015293/workspace.zip)

stewartoallen, 2021-08-19:
usually terrain like this would be `contour` operation

geobruce, 2021-08-21:
We tried a few different finishes on a small landscape but we'd prefer to have it with a waterline finish.
In the end we'd like to make a rather big landscape 1m x 2m. For our end product we would like for example a "rough" finish operation with step down 3mm. But as we don't want to clear all unneeded material with 3mm step down we would like to clear it with a step down of 8mm and then only clear the material that is left with a step down of 3mm. 

just to illustrate our experiments:
[image]
[image]

stewartoallen, 2021-08-23:
I think using the `clip to` option on stock may allow you to do what you want with outlining. should solve the red circle annotation from your earlier post.

[image]

geobruce, 2021-08-23:
> I think using the `clip to` option on stock may allow you to do what you want with outlining. should solve the red circle annotation from your earlier post.
> 
> [image]

Thanks I tested this and it works!

geobruce, 2021-08-23:
> usually terrain like this would be `contour` operation

We tested a few strategies and we really like do the look of the rough operation.
Two problems with this:

1. If there is no option for **rest machining** this will be incredibly in efficient.
2. With clear faces enabled in rough, it slows the pocketing down with a whole layer on that height that will be surfaced instead of just the face.

[image]

[image]

[image]

[image]
[image]
[image]
[image]
[image]

stewartoallen, 2022-10-09:
This is what I would do in 3.4+ -- it's outline clipped to stock

[outline-clip-to-stock.kmz.zip](https://github.com/GridSpace/grid-apps/files/9742519/outline-clip-to-stock.kmz.zip)

[image]

closed 2022-10-09 by stewartoallen