GridSpace/grid-apps #187: im not able to run the project

maher610, 2021-09-02:
i followed all the read me steps and when i open local host im getting "cannot get /kiri" can anyone please help im still new to Node.js

stewartoallen, 2021-09-07:
Are you on Windows?

maher610, 2021-09-12:
yes

stewartoallen, 2021-09-12:
what about this part of the readme ...

**Windows Developers**
this git repo requires symbolic link support. on Windows, this means you have to clone the repo in a command shell with Administrator privileges.

maher610, 2021-09-12:
i tried opening windows powershell as an administrator and followed the steps but i get the same error

stewartoallen, 2021-09-12:
have you tried WSL? this issue only exists on Windows. there are others using Windows as their primary platform, so perhaps ask around on the forum or Discord server.

maher610, 2021-09-12:
i will try that now, Thanks for helping

closed 2021-11-28 by stewartoallen