GridSpace/grid-apps #189: slicer engine CNC example

frank-pian, 2021-09-27:
Can you show me an example of slicer engine CNC ?
This one doesn't work

```
kiri.newEngine()
                .setListener(display_message)
                .setMode("CAM")
                .load("/obj/cube.stl")
                .then(eng => eng.setProcess({
                    camRegisterSpeed: 1000,
                    camRoughDown: 3,
                    ​​camRoughFlat: true,
                    ​​camRoughIn: false,
                    ​​camRoughOn: true,
                    ​​camRoughOver: 0.4,
                    ​​camRoughPlunge: 250,
                    ​​camRoughSpeed: 1000,
                    ​​camRoughSpindle: 1000,
                    ​​camRoughStock: 0,
                    ​​camRoughTool: 1001,
                    ​​camRoughTop: true,
                    ​​camRoughVoid: true,
                }))
                .then(eng => eng.slice())
                .then(eng => eng.prepare())
                .then(eng => eng.export())
                .then(display_gcode);
```

stewartoallen, 2021-09-27:
Digging into this a little, there is development to be done for CAM mode to work properly. It'll require settings stock, tools, and other parameters that the UI does automatically. Scoping the effort now.

stewartoallen, 2021-09-27:
I've made changes to allow this code to work with the next code push to production:

```
kiri.newEngine()
    .setListener(display_message)
    .setMode("CAM")
    .load("/obj/cube.stl")
    .then(eng => {
        eng.setTools([{
            type: "endmill",
            number: 1000,
            metric: false,
            diameter: 0.25,
            flute_diam: 0.125,
            flute_len: 0.5,
            shaft_diam: 0.125,
            shaft_len: 0.75
        }]);
        eng.setDevice({
            deviceName: "test",
            gcodePre: [ ";; header "],
            gcodePost: [ ";; footer "],
            gcodeSpace: true,
            gcodeString: false
        });
        eng.setProcess({
            camRegisterSpeed: 1000,
            camRoughDown: 3,
            camRoughFlat: true,
            camRoughIn: false,
            camRoughOn: true,
            camRoughOver: 0.4,
            camRoughPlunge: 250,
            camRoughSpeed: 1000,
            camRoughSpindle: 1000,
            camRoughStock: 0,
            camRoughTool: 1001,
            camRoughTop: true,
            camRoughVoid: true,
            ops: [{
                type: "rough",
                tool: 1000,
                leave: 0,
                down: 1,
                rate: 1000,
                plunge: 250,
                step: 0.5,
                top: true,
                flats: false,
                inside: false
            }]
        });
        return eng;
    })
    .then(eng => eng.slice())
    .then(eng => eng.prepare())
    .then(eng => eng.export())
    .then(display_gcode);
```

to get more information on what's expected in these data structures (tools, device, etc) look in the [conf.js](https://github.com/GridSpace/grid-apps/blob/master/src/kiri/conf.js) file or from the JS console with Kiri:Moto open in the browser, use the `kiri.api.conf.get()` function to get a navigable object:

[image]

stewartoallen, 2021-09-27:
fixed in https://github.com/GridSpace/grid-apps/commit/7ea9f2a7ef58c68eb3f5e81ef00fea777bf34151

closed 2021-09-27 by stewartoallen