GridSpace/grid-apps #191: Toolpath bug V3.0 and 3.1

geobruce, 2021-12-02:
[image]
This bug is apparent in v3.0 and 3.1
[rough bug workspace.zip](https://github.com/GridSpace/grid-apps/files/7641893/rough.bug.workspace.zip)

I didn't test older versions.

stewartoallen, 2021-12-02:
it does a little better than 2.9 -- but I noticed some mesh defects .. the new "heal" operation does an integrity check and noticed gaps. and I found these tags on the model

[image]

[image]

geobruce, 2021-12-02:
Is the heal test also able to fix this problem?

[Screen Shot 2021-12-02 at 5 15 20 PM]<https://user-images.githubusercontent.com/554275/144512096-b869f286-13a7-47c8-8d69-5b2bb48a5429.png>

[Screen Shot 2021-12-02 at 5 15 27 PM]<https://user-images.githubusercontent.com/554275/144512107-8d91a290-0e38-4ec0-a530-2aaed0f08a06.png>

stewartoallen, 2021-12-02:
unfortunately, no. I haven't gotten eyes on it yet. it could be minute. all I know is that there are open edges somewhere. they could be holes or overlapping faces, or both.

stewartoallen, 2021-12-02:
the defects in the model line up with the tabs pictured above. it's a series of overlapping and open faces that are very close to the faces of these recesses in the model. this confuses the slicer where the defective faces intersect the model.

[image]

stewartoallen, 2021-12-08:
I imported the STL into Onshape and it reports the same detached bodies. I would like to verify the toolpaths on a clean model. Are you able to produce one?

stewartoallen, 2022-10-09:
I'm closing this as a non-manifold / fuzzed model issue.

closed 2022-10-09 by stewartoallen