GridSpace/grid-apps #192: CLI not working

Lenergyy, 2022-01-07:
I am trying to use the CLI to automate the some CNC toolpath generation.
I used [this ](https://gitmemory.cn/repo/GridSpace/grid-apps/issues/179) thread as a guid.
But when i try to replicate it, I just get a bunch of errors:

[image]

I am using the latest versions of node and npm.
It would be very nice if someone would take the time to help me troubleshoot the problem.

stewartoallen, 2022-01-07:
try the 3.0 branch for stability. I'm making a ton of internal changes in 3.1 that likely caused this. the fix should be easy, though.

stewartoallen, 2022-01-07:
it's great to have this memorialized in the issue tracker so I don't forget to patch it up before 3.1 final. but it you want real-time help, hope on over to the Discord server or the forum. Both linked from the bottom of the grid.space home page.

commit 63a71d07 (2022-01-07): fixes #192 cli broken
closed 2022-01-07 by stewartoallen