GridSpace/grid-apps #195: snapmaker (origional) support?

JamesRH, 2022-01-17:
As a follow-up to a forum posting here: https://forum.grid.space/t/snapmaker-2-profile-s-request-fdm-laser-cnc/295

To quote the original poster:
> SnapMaker...is a combo FDM/Laser/CNC modular unit that uses Marlin-esq gCode for FDM and a custom version of GRBL for both the laser and CNC functions. If I can provide sample output code for each module, can this machine be integrated into K:M?

The snapmaker original is a $600 multi-machine whose provided software is very limited. Seems like a great fit for bringing a large user base to K:M!

The info I posted from the forum linked to above:
from https://snapmaker.com/snapmaker-original

3D PRINTING
Build Volume 125 x 125 x 125mm

LASER ENGRAVING
Work Area 125 x 125 mm

CNC CARVING
Work Area 90 x 90 x 50 mm

Here are some links to open source code from the manufacturer:

CNC post-processing plugins for FreeCAD and others, and gcode info here:
https://github.com/Snapmaker/snapmaker_cnc_post_process

Custom changes to Cura and gcode start / end for 3d slicing here:
https://github.com/Snapmaker/Snapmaker-3D-Printing-Profiles

Example gcode for 3dprinting, laser, and cnc available here (generated with snapmaker luban 4.1.3 and mostly normal settings):
https://github.com/JamesRH/snapmaker/tree/main/example_gcode

Let me know if I can be of any help in testing.

   James H.

stewartoallen, 2022-01-19:
Thanks @JamesRH .. I'll get profiles integrated before the 3.1 release this weekend. And I'll update here when they're live in the 3.1 beta if you want to test.

stewartoallen, 2022-01-19:
I'm going to push these to 3.1 production later today. Do you happen to know if the coordinate system on the Snapmaker uses the corner as origin or center of the bed as origin (0,0) ?

JamesRH, 2022-01-19:
The corner (lower left), based on the following:

In 3d printing mode it homes to the closest left corner (not used in laser and CNC mode as the work origin is set as part of the workflow). 

Entering the following gcode after turning on (with firmware: snapmaker-luban 4.1.3):

```gcode
G28 ;home
    ok
G90 ;absolute positioning
    ok
G1 X0 Y0 Z0
    ok 

;moves to lower left from homing position just below that

M114; Get Current Position
    X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0 Y:0 Z:106
    ok

```
So, looks like lower left (if I have my gcode commands correct). 

Also, some additional info that may be of use:

```gcode

M503 ;Get Current Settings
    echo:Steps per unit:
    echo:  M92 X400.00 Y400.00 Z400.00 E92.60
    echo:Maximum feedrates (mm/s):
    echo:  M203 X300.00 Y300.00 Z5.00 E25.00
    echo:Maximum Acceleration (mm/s2):
    echo:  M201 X1000 Y1000 Z100 E10000 X_CNC100 Y_CNC100 Z_CNC100 E_CNC100 X_LASER3600 Y_LASER3600 Z_LASER3600
    echo:Accelerations: P=printing, R=retract and T=travel
    echo:  M204 P1000.00 R1000.00 T1000.00
    echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
    echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
    echo:Home offset (mm):
    echo:  M206 X-4.50 Y-3.50 Z-3.06
    others < Mesh bed leveling:
    echo:  M420 S1 X2 Y2
    echo:  M421 X10.00 Y10.00 Z0.24
    echo:  M421 X115.00 Y10.00 Z-0.06
    echo:  M421 X10.00 Y115.00 Z0.04
    echo:  M421 X115.00 Y115.00 Z-0.21
    echo:Z2 Endstop adjustment (mm):
    echo:  M666 Z0.00
    echo:PID settings:
    echo:  M301 P38.00 I25.00 D300.00 C100.00 L20
    echo:Filament settings: Disabled
    echo:  M200 D3.00
    echo:  M200 D0
    ok

M119; Get EndStop Status
    others < Reporting endstop status
    others < x_min: open
    others < y_min: open
    others < z_min: open
    ok

```

stewartoallen, 2022-01-20:
new profiles pushed to 3.1 production. interested to know how they work for you.

JamesRH, 2022-01-24:
Tested with live version at https://grid.space/kiri/ version 3.1.1 with Snapmaker Original machine, firmware snapmaker-luban 4.1.3

Snapmaker Original machine loads in FDM, laser, and CNC settings and generates plausible gcode for all 3.  FDM 3d printing gcode executed correctly, locations correspond to print volume and filament advancing seems about right. CNC gcode checks out in CAMotics preview, but I need to confirm tool sizes. 

Thanks! I think this issue could be closed out. 

However, with default settings in the 3d test cube there was very poor first layer adhesion. I'll take a closer look and put in a pull request for tweaks to the settings and a few changes to the start and end gcode to match the behavior with cura. I'll try to test the laser engraving and CNC operation on real material next weekend.

stewartoallen, 2022-01-24:
that's great to hear. thanks for the follow-up. I look forward to any further feedback you may have.

closed 2022-01-24 by stewartoallen