GridSpace/grid-apps #198: [Feature] SLA Resin Volume

503Dev, 2022-01-31:
Hello,

I am working with the latest version of Kiri:Moto on a local dev server.  I noticed SLA operations do not return the volume of estimated resin usage. 

Is there presently any method to obtain this programmatically from the slicing functionality?

stewartoallen, 2022-01-31:
added volume to export in this commit: https://github.com/GridSpace/grid-apps/commit/0b19a6c98f4c575fbe6759427311eca112e73c61

503Dev, 2022-01-31:
Thank you, you are a legend and this project is a godsend for the 3d printing community.

closed 2022-01-31 by stewartoallen

503Dev, 2022-01-31:
@stewartoallen Just a question (sorry to nag at your time):

What is the unit being returned? I applied the changes and sliced the test cube and it returns a value of 7200 which is now what I am normally used to seeing.  I might just be missing something, thanks again!

stewartoallen, 2022-01-31:
It should be cubic millimeters. what would you expect?

503Dev, 2022-01-31:
Sorry I am ridiculous, I have been working with other resin slicers and they always display ML which is what I was thinking, cubic mm is obviously what should be expected problematically.  My mind is in the code gutter.  Thank you again.

stewartoallen, 2022-01-31:
happy to make that conversion (just 1/1000) since you're the only one asking :)

503Dev, 2022-01-31:
:) thank you.  Seriously, I am indeed thankful for your awesome work and contributions.