GridSpace/grid-apps #199: kiri:moto cli.js produces 144mm offset from Z0 (on belt printer)

smartin015, 2022-02-11:
I have a CR-30 belt printer, and I'm using [cli.js](https://github.com/GridSpace/grid-apps/blob/master/src/kiri/cli.js) with some small tweaks to slice my files from the command line. Pretty much all of the tweaks are to facilitate loading in the settings files.

I can successfully slice an STL, but when I run the output, I expect to start printing near where the belt is currently positioned. What I see instead is a Z-offset of ~144mm on the first layer after the purge line. In effect, this rolls the belt for 144mm before it starts printing, which causes it to print on a cooler portion of the bed (leading to adhesion issues)

trying `engine.moveTo(0,0,-144)` doesn't seem to have any effect, nor does `engine.center` of the same.

Device config: https://pastebin.com/AcZ9VMfD
Slicer config: https://pastebin.com/dEcmgf10
My `cli.js`: https://pastebin.com/BQL7EMg4
Example gcode: https://pastebin.com/6iD5aqt6 (see line 110 - moves to Z144.5326)

I tried to copy my settings from the web instance of kiri:moto. Maybe I'm missing a setting somewhere to get it to print at the start? Thanks in advance!

stewartoallen, 2022-02-11:
obviously something unexpected is happening here. I'll track it down and update here when it's fixed.

side note: seems like the cli is getting a lot more attention. I'm going to formalize that into a more functional parameterized utility instead of just sample code.

smartin015, 2022-02-11:
Wasn't obvious to me :) 

Despite the weird difference in behavior, I still use it pretty often - thanks for all the work you've put into it so far.

stewartoallen, 2022-02-11:
solved. the confusion is the Y/Z axis swap in belt mode. the API still uses Y, so you want to move that axis, not Z. using the new cli tool that I just committed and the device/process files you pasted above, and this command line
```
node src/kiri/cli --device=/src/cli/belt-device.json --process=/src/cli/belt-process.json --output=/tmp/belt.gcode --move=0,-130,0
```
I get the correct output
[image]

closed 2022-02-11 by stewartoallen

wjsteele, 2022-02-11:
Just a side note, on some printers, the X is the long axis as well... and on other's, it's the Z.  When I first developed the belt printers, the original was based off the Makerbot Replicator, where the Z was turned into the belt... and that convention kind of stuck for most printers... however, with the Blackbelt style of system, it didn't make sense for some designers that the flat belt was still the Z, so they flipped it... and made the angle the Z axis and the belt have the bottom of the part based on the X/Y axis... and, like on some Printrbot Printrbelts, the were even flipped... so the long axis was X and the side axis was Y.

By the way... Amazing job on all of this!  It's great to see the belt printers taking off.  They're so much more than I ever anticipated.