GridSpace/grid-apps #202: after manual layout, adding a new object reverts to auto-layout; should keep manual layout

tschundler, 2022-03-29:
Unfortunately, without #49, it's necessary to layout manually in Kiri:moto. But if I have a layout and want to add something since I have space, it resets the layout.

If layout is manually adjusted, it should set a "dirty" bit for the layout. If that bit is cleared, when a new object, it should not move the existing ones. The flag gets reset by clicking the autolayout option or clearing all objects.

stewartoallen, 2022-03-29:
Layout will only happen when objects are added if `auto layout` is enabled.
[image]

tschundler, 2022-03-30:
Thanks, that solves my problem.

I still think automatically switching would still be the best UX, but feel free to close this if you think that feature request isn't practical.

stewartoallen, 2022-03-30:
great. I will close this for now.

closed 2022-03-30 by stewartoallen