GridSpace/grid-apps #207: Climb vs conventional ignored on layer with tabs in "outline" mode for CNC mill

tschundler, 2022-06-07:
This can be seen in the workspace in #206.

What happens:
When tabs are present, the initial layers are cut in the climb or conventional direction per settings. But on the tab layer, when it encounters a tab, sometimes it changes cutting direction. On my CNC router, that makes a notable difference in finish.

What I expect:
When tabs are encountered, it should raise z to the tab height (G0 or G1) then clear the tab (G1... or G0 at safeZ if tab height >= stock surface ) continuing in the same direction, then G1 plunge to the bottom and continue in the same direction.

stewartoallen, 2022-06-07:
this is a known issue. discussed on Discord but not documented here (until now). when the polygons are cut by tabs they "lose" their direction. mathematically you can't determine winding from an open loop. and the levels of abstraction in which the cuts occur obscure the original winding information. this will require a little work to resolve.

tschundler, 2022-06-07:
Ah, so it just becomes random segments? That would also explain the full Z-height raise instead of tab-height.

Having no idea what the code looks like (though maybe I'll take a look if I have time in the next month or so), I'd imagine the solution would be to process tabs after paths are generated rather than before. e.g. check intersections of the generated outline with the tabs (modeled as spheres or circles offset for simple point-line distance check if < tool radius) and on collision split the gcode segment.

stewartoallen, 2022-06-07:
it's a little more complex because of segment re-connect and routing logic. I imagine there is a way to tag the polygons that carries through. it would take a little more digging to see if this is feasible.