GridSpace/grid-apps #208: Artifacts on different slice height

p3d-dev, 2022-06-11:
Thanks for the great slicer. Still there are some issues. Here is one…

The slicing of this model without support works well at default slice height[cage in ball](https://www.thingiverse.com/thing:5269899/files). If sliced with 0.2mm, strange artifacts appear. This can be easily tried with the [version of kiri](https://grid.space/kiri/).
[image]

stewartoallen, 2022-06-12:
Thanks. I've repro'd the issue and will investigate. In the meantime, you can change the base layer size from 0.25 to anything else (like 0.24) and it will avoid the issue.

stewartoallen, 2022-10-04:
this workspace reproduces the issue. to help me repro again later. base layer height affects the outcome. [cage-and-ball-artifact.kmz.zip](https://github.com/GridSpace/grid-apps/files/9703047/cage-and-ball-artifact.kmz.zip)