GridSpace/grid-apps #209: Doubled extrusion

p3d-dev, 2022-06-13:
Hello again and sorry,

another issue can be seen with [3d benchy](https://www.thingiverse.com/thing:763622). Slicing with 0.19mm, there is a double extrusion visible on e.g. layer 186.
[image]

The red marked area indicates this phenomenon. I could not figure out, how to show the print head movement, which would make this double extrusion more obvious.

stewartoallen, 2022-06-13:
try enabling various thin wall algorithms under the layer menu to see if this helps

p3d-dev, 2022-06-13:
Thanks for the proposal. I will try this.

closed 2022-06-13 by p3d-dev