GridSpace/grid-apps #211: Kiri Moto Laser always at S255 despite 5 % setting

BKLronin, 2022-06-25:
Hi guys,

experimenting a lot with KM lately and CAM went well so far. Only problem Iǘe got and can´t find any clue is that the laser power is always set to S255. 5% or 100% different machine different settings. Its Always S255 in Gcode.

This is on Firefox, Manjaro.

Ayn hint would be of value. Laser is sitting ready for its first cut :).

stewartoallen, 2022-06-25:
I'll look into why the power value isn't being converted for `S{value}` properly ... in the meantime, you can hard-code the value in the device gcode macro

[image]

BKLronin, 2022-06-25:
Thank you very much. 

I ll try a fixed S value tomorrow.

BKLronin, 2022-08-25:
As I understand bitmaps or pixel based graphics would be lasered by altering the pwam for different intensieties or does it alter speed or number of times passing a layer?

stewartoallen, 2022-08-29:
Kiri:Moto does not raster images. Once sliced in laser mode, the depth data is gone.  So that's usually something left to programs like LightBurn.

BKLronin, 2022-08-29:
Ah thanks,

wasn´t sure if there is some raster mode that is triggered somehow. I had a SVG that contained a rasterized part and that information got missing on importing on most laser tools. 

So the fixed s value per machine setup is totally fine. Tried lightburn and it seems nice but I try tor reduce the nr of prgrams I use. 

BTW: Kiri is really nice, I do recommended it every now and then. (F360 cam replacement)

stewartoallen, 2022-08-29:
Great to hear. Always looking for feedback and ideas to improve it.

closed 2022-08-29 by stewartoallen