GridSpace/grid-apps #216: Kiri site runs out of memory when slicing a big object

miksch123, 2022-09-13:
I am accessing the kiri:moto slicer from Microsoft Edge on a Surface pro 6 and I cant get it to properly export a GCode. Everytime I do so the memory consumption goes up to 6 gigs of ram and the browser tab shows me a out of memory screen.
the model I tried to slice is: https://www.thingiverse.com/thing:2018105.
The stl used was: Cyclovac_with_thread.stl
the profile and workspace files are attached.
[orange amazon pla.km.txt](https://github.com/GridSpace/grid-apps/files/9555806/orange.amazon.pla.km.txt)
[workspace.kmz.txt](https://github.com/GridSpace/grid-apps/files/9555807/workspace.kmz.txt)

To use the files remove the .txt from the full file name.

stewartoallen, 2022-09-13:
Good news. I was able to replicate your OOM in Edge on Mac. And I was able to determine a way around it by disabling threading and reducing "detail" to 50. Threading increases memory pressure by trying to do more work at once. This can challenge the memory management of the JS VM. Also, the larger the model gets, the less detail is important. For a small benchy model, the difference between 50 and 100 might be subtly noticeable. For a large model like this, it just increases the size of the gcode (and memory and processing), but leads to no visible improvements.

[image]

closed 2022-09-13 by stewartoallen

miksch123, 2022-09-14:
thank you so much for that ^^