GridSpace/grid-apps #230: Tools with shared number can break simluation

stewartoallen, 2022-12-01:
by choosing the wrong tool

wcraigtrader, 2023-03-15:
I had defined both 1/4" tools and 1/8" tools, and I had given each tool a tool # to match their positions in the Carvera tool changer. This meant that I had a 1/4" spiral down cut bit with tool #1 and a 1/8" up cut end mill, each numbered as tool 1. (I actually had 6 numbered tools in each size.)

When animating a toolpath that was defined for the 1/8" end mill, the path was cutting with the 1/4" end mill properties.

Renumbering the 1/4" bits solved this problem, but it means that I'll have to renumber all the conflicting bits every time I switch collets and/or jobs.

stewartoallen, 2023-03-15:
I will revisit this for 3.9

commit fec2fcc1 (2023-03-17): fixes #230 shared tool numbers cause animation errors
closed 2023-03-17 by stewartoallen