GridSpace/grid-apps #239: API Not Working "https://grid.space/kiri/engine.html"

mahmood265, 2023-01-14:
Hi! I have checked the link that I have used to learn about it, and it is not working on webpage even I have cleared my cache. I am using edge.
Error: "i.client.unrotate is not a function"
To solve this issue I downloaded the file and run it on my server. I temporary removed it from engine.js file and try again. This time it works up to generating vertices and, stops working slicing forward. I have tried to figure it out but not found.
**Code I have used is same as described in link just changed used my own stl file
[image]
:**
```
kiri.newEngine()
    .setListener(display_message)
    .load("/obj/cube.stl")
    .then(eng => eng.setProcess({
        sliceShells: 1,
        sliceFillSparse: 0.25,
        sliceTopLayers: 2,
        sliceBottomLayers: 2
    }))
    .then(eng => eng.setDevice({
        gcodePre: [ "M82", "M104 S220" ],
        gcodePost: [ "M107" ]
    }))
    .then(eng => eng.slice())
    .then(eng => eng.prepare())
    .then(eng => eng.export())
    .then(display_gcode);

```

mahmood265, 2023-01-14:
[image]
Console output after temporary removing unrotate function.
Sorry to put it here it was not uploading in above document.

stewartoallen, 2023-01-15:
fixed. will be pushed to production shortly.

commit 9556e821 (2023-01-15): add CORS headers for SharedArrayBuffers. remove deprecated unrotate(). fixes #239. see https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/SharedArrayBuffer
commit 7686aeaf (2023-01-15): add CORS headers for SharedArrayBuffers. remove deprecated unrotate(). fixes #239. see https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/SharedArrayBuffer
closed 2023-01-15 by stewartoallen