GridSpace/grid-apps #241: Set acceleration; print speed visualization

vogler, 2023-01-27:
I like that the preview shows print speeds, but didn't find anything in the UI to change acceleration and somehow it showed the same speed for everything:

[image]

Does print speed mean the average speed for each move? I assume it wouldn't show a color gradient for a single move ([example curve](https://blog.prusa3d.com/calculator_3416/#speed)).

Here's what acceleration control in Cura looks like:

[image]

stewartoallen, 2023-02-01:
What is pictured is the speed communicated in gcode to the printer. What the printer actually does in terms of acceleration and deceleration is something that, unless explicitly set in the gcode, is not available to KM. What other slicers are likely doing is providing a helper based on printer firmware (accel/decel commands and settings differ between printers) to set those values in the gcode file being generated for a printer target.

As long as you're working with stock printers, this is usually a safe bet. If you're working with printers that have been modded and DIY printers or an unknown brand, this is not something I would be comfortable setting in gcode. The other issue with gcode overrides to firmware settings is that if one file sets it and another does not then override those settings, it's in effect from the time before. This can lead to unknown printer states.

Typically, I tune the accel/decel settings in firmware that works best for the printer and then leave them alone, not set them in the slicer.

All of this is possible in KM, but not feasible for one person. There are just too many printers to keep track of.

vogler, 2023-02-01:
Thanks, got it. So the shown speeds are mostly influenced by min. layer time?
I just happened to tune acceleration for my printer some time ago and have it increased a lot in my fast/low quality profile.
Do you know of any slicer that calculates and shows the real speed based on acceleration?

oliof, 2023-02-01:
The slic3r family of slicers (PrusaSlicer/SuperSlicer) and Cura assume accelerations based on printer profiles.  With the Printer Settings plugin, you can define acceleration values for custom printers for precise print time prediction without acceleration control from the slicer.

vogler, 2023-02-01:
Thanks, using Cura. Now I at least have a better time estimate (acceleration control doesn't affect it):
https://github.com/fieldOfView/Cura-PrinterSettingsPlugin
Had to set all of them:
[image]
[image]

vogler, 2023-02-01:
I'll close this then.

> The other issue with gcode overrides to firmware settings is that if one file sets it and another does not then override those settings, it's in effect from the time before. This can lead to unknown printer states.

You can't read the current value and then restore it at then end since there are no variables?
https://marlinfw.org/docs/gcode/M501.html will load all settings and you can't just load acceleration?

Yea, all pretty messy, maybe with a disclaimer.

closed 2023-02-01 by vogler

stewartoallen, 2023-02-01:
Slicers typically produce a stand-alone gcode file. Getting data from the printer requires an active network or serial connection. Also, those command are Marlin-specific and don't work with Klipper and other firmwares (they each have their own method)