GridSpace/grid-apps #243: Support importing STEP files in addition to STL files

wcraigtrader, 2023-03-13:
In the CNC world, STEP files are more common and can provide more precise definitions of a part (actual circles and curves instead of approximated circles and curves). Being able to import a STEP file directly instead of having to render it as STL first would be ideal.

stewartoallen, 2023-03-13:
There are a [few issues](https://discord.com/channels/688863523207774209/688863523211968535/1078662758578524170) with implementing STEP in KM. 

1. the step specification, while considered a "standard" is not free
2. the spec is quite extensive. coding up full support for it would be a years long project
3. unlike desktop software, I cannot just import libraries to handle this
4. KM's geometry engine does not natively support splines, so it would have to be faceted anyway

There are large / complex libraries that implement STEP and could be compiled to WASM. But they would only serve to convert STEP to STL or OBJ without a rewrite of KM's geometry engine.

I am open to any suggestions

oliof, 2023-03-14:
IMHO, internal conversion to STL/OBJ would be an acceptable first step to make design files printable, although I agree it's a neverending can of worms. The smallest STEP file interpreter implementation I know is formlabs' [foxtrot](https://www.mattkeeter.com/projects/foxtrot/) which is dual licensed under Apache and MIT. I am not sure if that already meets your definition of large/complex, but I wanted to drop a note on it here.

stewartoallen, 2023-03-14:
Thanks @oliof ... the foxtrot write-up is a gem. It's so much worse than I imagined (the spec thing). The resulting rust compiled to `.wasm` file is a 1.6M binary -- for reference, Kiri:Moto weighs in at 4MB with all assets included.

AleeeKoi, 2024-04-30:
Hello, I was looking for this feature and I just found this issue. Are there any updates?
It could be very useful, CNC world works with step files as source.
Thanks