GridSpace/grid-apps #258: Local deployment failed

linhuanfengQ, 2023-10-24:
After I run this command line, print the following output. Does this mean the startup was successful
gs-app-server
231024.095710 '[appserver]' 'app-server running: ports=[8080] apps=[apps] data=[data] logs=[logs]'

But when I visited the website, 404 appeared

stewartoallen, 2023-10-24:
windows? if so, you need to re-clone the repo w/ admin privileges so sym links work. if no, don't forget `/kiri` on the url

localhost:8080/kiri

linhuanfengQ, 2023-10-24:
[image]

Hello, my deployment on Centos also failed

linhuanfengQ, 2023-10-24:
Hello, may I ask if it is possible to update the deployment method, such as using NPM build and NPM start to achieve testing and production deployment, instead of using GS app server

stewartoallen, 2023-11-01:
happy to accept pull requests that address this

closed 2024-07-03 by stewartoallen